keith-packard/snek #6: A vote for LGPL

jockm, 2019-02-07:
Obviously you should use whatever license you prefer, but I would like to put a vote in for the Lesser GPL.  This would make it more attractive to embed Snek in a embedded app as a scripting/extension language.  I have done this on projects.  

Regardless, snek is very very cool.  Thank you.

keith-packard, 2019-02-07:
Thanks for your kind words. I'll stick with GPL though; I'm fine with requiring the entire source to a device being open.

jockm, 2019-02-07:
Fair enough, best of luck to you!

closed 2019-02-11 by keith-packard

vanillajonathan, 2021-12-14:
I also think the LGPL (or even the MIT License) would be a better choice than the GPL.

I think this ought to be reconsidered.

jockm, 2021-12-14:
@vanillajonathan For me it comes down to this: if I want to use snek in a product, I have to make my snek code GPL as well, because there is no linking exception (at least the last time I looked).  The FSF is pretty clear that if something is bundled in the same distributable and is wholly dependent on the GPL'd code, then it is part of the same source and covered by the GPL.  Snek is very cool and it would be nice to use in some embedded projects, and might get more people knowing about snek; but without a linking exception (at the very least) there is no way I can consider it

I am just sharing this so people can understand why someone might not think the GPL is the best choice for an embedded language (at least without a linking option).

vanillajonathan, 2021-12-14:
@jockm I see the same potential as you for Snek as an embedded scripting language, and I see the same issue as you with the license.