keith-packard/snek #8: bad input breaks state

fanoush, 2019-02-12:
when I press cursor up or enter invalid operator it break something and further lines are not evaluated properly

```
~/snek/posix$ ./snek
1+5
6
^[[A
<stdin>:2 Syntax error at "
+5
1+5
exit(0)

^C
~/snek/posix$ ./snek
1+5
6
1|3
<stdin>:3 type mismatch: <builtin (null)> <builtin (null)>
1+5
<stdin>:4 type mismatch: <builtin (null)> <builtin (null)>
exit(0)
~/snek/posix$
```
after cursor up I press enter key and enter 1+5 again and yet only +5 is shown
in second case the unknown operator | breaks evaluation but exit(0) works unlike the first case

keith-packard, 2019-02-12:
Thanks for your report -- the first issue was that the up-arrow command included a '[' character after the syntax error, which caused the lexer to switch back to 'ignore newline' mode (which lets expressions within parenthesis and square brackets span multiple lines). I've fixed that by resetting this value before each token is read during the syntax error recovery process.

As for your second issue; I suspect you've mis-copied the input -- the '|' operator is the bitwise OR operator, just as in Python. Let me know what character you used and I'll go explore what happens.

closed 2019-02-12 by keith-packard

fanoush, 2019-02-12:
yes, it was bitwise or and it didn't work. same with bitwise &. I thought both is not implemented as the only number datatype is float, if it should work then for me it doesn't. does the exactly same input I typed work for you?

fanoush, 2019-02-12:
oh, it is something else, the operator doesn't matter
```
~/snek/posix$ ./snek
1+5
6
1|3
<stdin>:3 type mismatch: <builtin (null)> <builtin (null)>
exit(0)
~/snek/posix$ ./snek
1|3
3
1+5
<stdin>:3 type mismatch: <builtin (null)> <builtin (null)>
exit(0)
~/snek/posix$ ./snek
1&3
1
1+3
<stdin>:3 type mismatch: <builtin (null)> <builtin (null)>
exit(0)
```

fanoush, 2019-02-12:
oh, sorry, forget it, it was my change. I changed snek_poly_t to struct just to see how passing by value works if it is larger than single value, so .u and .f was not same location

keith-packard, 2019-02-12:
cool; glad you're having fun!