keith-packard/snek #10: avr-gcc dependencie

eamanu, 2019-02-14:
avr-gcc is required right? I would be great that avr-gcc is listed on dependencies section.

eamanu, 2019-02-14:
After installation of avr-gcc I have the next problem:

```
make[1]: Leaving directory '/home/eamanu/dev/snek/posix'
make[1]: Entering directory '/home/eamanu/dev/snek/snek-duino'
lola -o snek-gram.h ../snek-gram.ll
multiple productions match ('MINUS', 'expr-add-p') - () and ('MINUS', '@ goto binop_first; ', 'expr-mul', '@ goto binop_second; ', 'expr-add-p')
multiple productions match ('PLUS', 'expr-add-p') - () and ('PLUS', '@ goto binop_first; ', 'expr-mul', '@ goto binop_second; ', 'expr-add-p')
multiple productions match ('OP', 'expr-array-p') - () and ('OP', '@{\n\t\t\t\tsnek_code_set_push(snek_code_prev_insn());\n\t\t\t}', 'opt-actuals', 'CP', '@{\n\t\t\t\tsnek_code_add_op_offset(snek_op_call, value_pop().offset);\n\t\t\t}', 'expr-array-p')
multiple productions match ('OS', 'expr-array-p') - () and ('OS', '@{\n\t\t\t\tsnek_code_set_push(snek_code_prev_insn());\n\t\t\t}', 'array-index', 'CS', '@{\n\t\t\t\tbool slice = value_pop().bools;\n\t\t\t\tif (slice) {\n\t\t\t\t\tsnek_code_set_push(snek_code_prev_insn());\n\t\t\t\t\tbool stride = value_pop().bools;\n\t\t\t\t\tbool end = value_pop().bools;\n\t\t\t\t\tbool start = value_pop().bools;\n\t\t\t\t\tsnek_code_add_slice(start, end, stride);\n\t\t\t\t} else {\n\t\t\t\t\tsnek_code_add_op(snek_op_array);\n\t\t\t\t}\n \t\t\t}', 'expr-array-p')
multiple productions match ('STRING', 'strings-p') - () and ('STRING', '@{\n\t\t\t\tsnek_code_set_push(snek_code_prev_insn());\n\t\t\t\tsnek_code_add_string(snek_token_val.string);\n\t\t\t\tsnek_code_add_op(snek_op_plus);\n\t\t\t}', 'strings-p')
python3 ../snek-builtin.py ../snek-keyword.builtin ../snek-base.builtin snek-duino.builtin -o snek-builtin.h
  CC  snek-builtin.o
In file included from ../snek-builtin.c:15:0:
../snek.h:17:19: fatal error: stdio.h: No such file or directory
compilation terminated.
../snek.defs:83: recipe for target 'snek-builtin.o' failed
make[1]: *** [snek-builtin.o] Error 1
make[1]: Leaving directory '/home/eamanu/dev/snek/snek-duino'
Makefile:29: recipe for target 'all' failed
make: *** [all] Error 2
```
But the snek on posix seems ok

keith-packard, 2019-02-14:
I neglected to indicate that you also need the avr-libc package installed, which is where stdio.h comes from. I've expanded the set of dependencies required and included the list of Debian packages you would need to build and use Snek. Let me know if that helps.

eamanu, 2019-02-15:
Perfect! thanks!

closed 2019-02-15 by eamanu