keith-packard/snek #9: make problems

eamanu, 2019-02-14:
Hi, 

I try to make snek. But when I run `make` I have this problem:

```
for dir in posix snek-duino; do (cd $dir && make); done
make[1]: Entering directory '/home/eamanu/dev/snek/posix'
lola -o snek-gram.h ../snek-gram.ll
make[1]: lola: Command not found
../snek.defs:62: recipe for target 'snek-gram.h' failed
make[1]: *** [snek-gram.h] Error 127
make[1]: Leaving directory '/home/eamanu/dev/snek/posix'
make[1]: Entering directory '/home/eamanu/dev/snek/snek-duino'
lola -o snek-gram.h ../snek-gram.ll
make[1]: lola: Command not found
../snek.defs:62: recipe for target 'snek-gram.h' failed
make[1]: *** [snek-gram.h] Error 127
make[1]: Leaving directory '/home/eamanu/dev/snek/snek-duino'
Makefile:29: recipe for target 'all' failed
make: *** [all] Error 2
```

This is because I don't have the dependencie `lola`. But lola make generate some files and binary but, snek README.md don't say me where copy lola on snek.

I think that a good improve on README file is write something about where lola go

keith-packard, 2019-02-14:
I've updated the lola project to add an 'install' target to the Makefile, and included instructions for downloading and installing that in the Snek README.md file. Let me know if those look good to you.

eamanu, 2019-02-15:
Work perfect! thanks!

closed 2019-02-15 by eamanu