keith-packard/snek #15: Doc doesn't build

BartMassey, 2019-06-08:
Debian with `asciidoctor` and `ruby-asciidoctor-pdf` installed. Output of `make` in `docs` directory:

```
asciidoctor-pdf -B.. --attribute="revdate=2019-05-19" --attribute="version=0.98" --attribute='pdf-style=snek'  snek.adoc
/usr/lib/ruby/vendor_ruby/pdf/core.rb:12:in `require_relative': /usr/lib/ruby/vendor_ruby/pdf/core/page.rb:56: unknown type of %string (SyntaxError)
        resources[:ProcSet] = %i[PDF Text ImageB ImageC ImageI]
                                 ^
/usr/lib/ruby/vendor_ruby/pdf/core/page.rb:56: syntax error, unexpected $end
        resources[:ProcSet] = %i[PDF Text ImageB ImageC ImageI]
                                 ^
	from /usr/lib/ruby/vendor_ruby/pdf/core.rb:12:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/vendor_ruby/prawn.rb:7:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/vendor_ruby/asciidoctor-pdf/converter.rb:3:in `<top (required)>'
	from /usr/lib/ruby/vendor_ruby/asciidoctor-pdf.rb:3:in `require_relative'
	from /usr/lib/ruby/vendor_ruby/asciidoctor-pdf.rb:3:in `<top (required)>'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/lib/ruby/1.9.1/rubygems/custom_require.rb:36:in `require'
	from /usr/bin/asciidoctor-pdf:6:in `<main>'
make: *** [Makefile:29: snek.pdf] Error 1
```

keith-packard, 2019-06-09:
That's odd. You might try with all of the package build dependencies that debian declares installed?

Build-Depends: debhelper (>= 11), libreadline-dev, avr-libc, gcc-avr,
 binutils-arm-none-eabi, gcc-arm-none-eabi, libnewlib-nano-arm-none-eabi,
 python3, lola, gawk, asciidoctor, ruby-asciidoctor-pdf, ruby-coderay

Hrm. I don't have ruby 1.9.1 installed, but have 2.5.1 instead -- I bet that's the problem. If so, I need to update the requirements for ruby-asciidoctor-pdf.

keith-packard, 2019-06-09:
Ok, problem resolved -- Bart had a debian 'alternative' redirecting 'ruby' to version 1.9.1. Fixing that solved this issue.

closed 2019-06-09 by keith-packard