keith-packard/snek #17: Incorrect handling optional parameters

phsilva, 2019-06-24:
Snek is currently not tracking the number of named parameters correctly. See some cases below:

Not sure if that is totally needed but would be nice probably. Let me know if you think this should be fixed and I can come up with a patch.

```
Welcome to Snek version 1.0
> def f(x=0): print(x)
>
> f()
0
> f(1)
<stdin>:4 wrong number of args. wanted 0 got 1
>

Welcome to Snek version 1.0
> def add(x=0, y): print(a,b)
>
> add(1,2)
<stdin>:3 wrong number of args. wanted 1 got 2
>
```

Second case is not allowed in cPython because of the ambiguity on it.

```
>>> def add(x=0, y): print(a,b)
...
  File "<stdin>", line 1
SyntaxError: non-default argument follows default argument
```

keith-packard, 2019-06-24:
> Snek is currently not tracking the number of named parameters correctly. See some cases below:

These are nice edge cases, thanks!

> Not sure if that is totally needed but would be nice probably. Let me know if you think this should be fixed and I can come up with a patch.

The closer we can make it to python, the better.

> > def f(x=0): print(x)
> >
> > f()
> 0
> > f(1)
> <stdin>:4 wrong number of args. wanted 0 got 1

snek does not (currently) let you pass positional actuals to named formals. You have to call this with f(x=1).

> > def add(x=0, y): print(a,b)
> >
> > add(1,2)
> <stdin>:3 wrong number of args. wanted 1 got 2

Same reason as above -- the actual needs to use the named form. The grammar allows named formals in any position, but requires named actuals to appear at the end. This fails to match python *and* is self-inconsistent. Note that you can use this form like this:

    > def add(x=0,y): return x+y
    > add(1)
    1
    > add(1,x=2)
    3

But, allowing named formals to appear before the last positional formal is definitely a bug.

keith-packard, 2019-06-25:
Ok, exploring this uncovered a nice set of bugs in the parser related to using the 'actuals' grammar bits for all sorts of things, including the range operator, list immediate values and tuple immediate values.

I fixed those bugs along with checking formals to make sure positional formals are not listed after named formals. I haven't figured out how I might make positional actuals match named formals yet; that looks hard. At least snek doesn't crash when you type things like [x=1] now.

phsilva, 2019-06-26:
Awesome @keith-packard. Will take a look at it too.

keith-packard, 2019-06-26:
I was reading through the Python3 spec about where the values for formals are evaluated -- it turns out Python3 says they're evaluated at the time of the function definition, not when the function is called. Snek evaluates them during the function execution (and they get evaluated even if not used). I'll update the docs to describe the precise semantics of the current implementation.

keith-packard, 2019-06-27:
Well, that took a bit of work. Check out the commit message to see how I understand the differences between actuals and formals in Python. Along with the fix for this, I've added a pile of tests to verify that snek now matches Python.

closed 2019-06-27 by keith-packard

rofl0r, 2019-06-27:
for the readers convenience, this is the commit in question https://github.com/keith-packard/snek/commit/ad2e91df43b94632c48230c98814627add98b80e