keith-packard/snek #26: [port] 1284p Snek pit?

drawkula, 2019-11-20:
Are there hints somewhere how to get Snek into an ATmega1284p?
That breadboard friendly DIP40 with lots of RAM cutie would make a nice Snek pit. ;-)

keith-packard, 2019-11-20:
Yes indeed! Should be pretty easy to make this work; there are examples for the 328 and 2560. I'd start with the 'mega' port and fix the Makefile to adjust the -mmcu compiler flag along with the LDFLAGS values which represent how much memory of each type the device has. After that, it's a more complicated matter of adjusting the GPIO code to match the device.

Note that Snek requires a serial port with XON/XOFF flow control, which means either an FTDI chip or hacked firmware if the board is using at atmega16u2 for USB to serial conversion.

keith-packard, 2019-11-20:
We can leave this open and see if there is more interest in porting to this target. Do you know of any pre-built board that has this chip on it? I bet we can find someone interested in doing the software part if there's an easier hardware platform to start with?

drawkula, 2019-11-21:
The hardware part is the easier 10%.

* The Goldilocks board probably is the most famous Arduino sized 1284P. It uses an ATmega32u2 instead of the ATmega16u2. But I don't know when/whether it will be restocked: https://www.freetronics.com.au/products/goldilocks-arduino-compatible-with-atmega1284p-mcu
* A minimal 1284 breadboard setup is explained in: http://www.technoblogy.com/show?19OV
* The 1284 should be compatible with these "minimal ATmega16 development boards", so just a FTDI cable or board would be needed for the serial connection and the instructions above will fit too: https://www.ebay.ca/itm/201622043605

The additionally needed ISP (or a 2nd Arduino as ISP) would be no hurdle too.

keith-packard, 2020-01-30:
Looks like crowd supply has an upcoming campaign to build a 1284-based board.

[644/1284 Narrow](https://www.crowdsupply.com/pandauino/narrow)

This would be a good target for Snek, when it becomes available.

keith-packard, 2021-03-23:
I've gotten one of these boards and snek now supports it!

closed 2021-03-23 by keith-packard