keith-packard/snek #28: Any chance to port Shek to BBC:Microbit or ESP*?

tangingw, 2019-12-02:
Is there any chance that Snek will be ported to BBC:Microbit or ESP(8266|32)? 

Can we do cross-compiling to other platforms?

keith-packard, 2019-12-02:
I don't have either of those two platforms to do the work. But, I'd love to help if you are interested in giving either one of these a shot.
Snek is cross-compiled for all of the embedded platforms. I do all of my builds on Debian as there are packaged compilers for RISC-V, ARM and AVR. I've recently switched to using picolibc for RISC-V and ARM, which isn't in Debian yet, but can be built fairly easily.

tangingw, 2019-12-02:
@keith-packard Thanks for the prompt response.

I believe your toolchain might have compiled snek to all ARM flavours, hence theoretically it should work on Microbit since Microbit runs on ARM Cortex-M0. ESP related might be another beast to tackle since they have different instruction sets compared to ARM. 

Speaking of snek, I find that it is fairly easy for a child to pick up within a short time. A side line question: will snek allow us to create our own modules for snek?

keith-packard, 2019-12-02:
Yes, porting to the Microbit should be pretty straightforward. You will need a UART driver, code to initialize the clocks and a driver for the GPIO pins and whatever other devices are on the board. I had considered starting this port, but I'm focused on teaching with snekboard and Lego robotics this term.

As for the ESP8266 and ESP32 chips; the level of effort is about the same as there is already a C compiler and library available for both devices. 

The real work is in the drivers for the particular target SoC. This can be as little as a few hundred lines of code, like the support for ATmega parts in snek which only needed a UART, eeprom and GPIO drivers. Or it might require a few thousand lines, like the SAMD21 support which includes a USB driver and some rather complicated clock configuration logic.

I think by 'snek modules' you mean some way of splitting code out into libraries for re-use. I'm not sure how this might work on the device as there isn't any file system on the target device, only space for a single snek script which is executed at boot time. Having only a single script on the device greatly simplifies the system as there isn't any question of what program to load and run. If you wanted to have a mechanism inside the 'mu' editor for pulling in blocks of code before sending them to the device, I can definitely see how that might be useful.

tangingw, 2019-12-03:
@keith-packard, 

Thanks for replying. For the modules part, I am thinking something that is rather wild. Correct me if I am wrong on this matter: Let's say for example if I want to call the methods from requests (a popular module in Python), is it possible for me to port this into snek? My understanding on the structure of snek is it might not be since it is designed for a 32k system.  However would be possible for us to build something that is modular enough to add into snek?

For the UART, EEPROM and GPIO, I might cross-reference how Micropython does it. Perhaps it might shed some light for me. A little note for myself: Intermediate Python user, shallow knowledge on C, but interested with how small embedded system works. 

I would definitely like to contribute to this project.

keith-packard, 2019-12-03:
>  Let's say for example if I want to call the methods from requests (a popular module in Python), is it possible for me to port this into snek?

I went and looked at this module; it needs all sorts of stuff not available in snek -- the most obvious being 'objects', which snek doesn't support at all.  Not to mention OS support for networking...

> My understanding on the structure of snek is it might not be since it is designed for a 32k system.

Snek is a very tiny language compared to Python. My goal is to make something that will serve as a good introduction to programming while also giving them tools to help learn python in the future.

> However would be possible for us to build something that is modular enough to add into snek?

You can certainly write functions for snek that would be useful in many programs; right now there isn't any way to 'import' things into the environment as that requires a place to store things by name.
 
> For the UART, EEPROM and GPIO, I might cross-reference how Micropython does it. Perhaps it might shed some light for me. A little note for myself: Intermediate Python user, shallow knowledge on C, but interested with how small embedded system works.

A good place to start learning embedded programming might be to get one samd21 boards from adafruit that snek already runs on. Building snek from source for that would expose you to quite a lot of how embedded development is done.

> I would definitely like to contribute to this project.

I ordered a ESP-WROVER-KIT from mouser today; it looks like getting snek running should mostly be a matter of copying from other existing projects. I'll see if  I can't get the basic system running. If you're interested in helping out, there will be plenty of opportunities to play there.

tangingw, 2019-12-03:
Hi @keith-packard,

I found some clones of SAMD21 from my local stores. I wonder if snek works on [first clone](https://www.lazada.com.my/products/samd21-m0-mini-32-bit-arm-cortex-m0-core-compatible-with-arduino-zero-form-mini-i484492862-s825966521.html) and [second clone](https://www.lazada.com.my/products/samd21-m0-32-bit-arm-cortex-m0-core-compatible-with-arduino-zero-arduino-m0-black-i206274070-s255949460.html)

keith-packard, 2019-12-03:
Those look like they're compatible with the Arduino Zero, which uses the same pinouts as the nano33iot board, which snek already runs on.

tangingw, 2019-12-04:
Thanks @keith-packard , will proceed with purchasing for the project.

keith-packard, 2019-12-05:
I received my ESP32-WROVER-KIT today and have pushed a very primitive port of snek to that to master. It does not build by default as it requires the ESP IDF to be installed. With that installed, it should be fairly easy to build and install on any ESP32 device.

It might be fun to extend that port to talk to the GPIO pins, or communicate over WiFi or Bluetooth; please feel free to give that a try.

I'll close this bug and see what people want to do from here.

closed 2019-12-05 by keith-packard