keith-packard/snek #35: Unable to connect to snekboard

tcharding, 2020-11-13:
Hi,

Not sure where to ask this question, mainly just looking for some tips on how to troubleshoot.

I received a snekboard a while ago to mess around with programming with my son.

I am using patched [mu-editor](https://github.com/keith-packard/mu) on Ubuntu 20.04

Attaching snekboard as directed in [lesson 1](https://sneklang.org/doc/lessons/lesson-1/lesson-1-snekboard.html#_mu_editor_mu_editor) gives me the following error:

```
Could not find attached device

Please make sure the device is plugged into this computer.

It must have a version of MicroPython (or CircuitPython) flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.
```

No amount of re-connecting or resetting gets me past this error.

The snekboard comes pre-flashed right? I don't need to flash it before first use?

Have you any suggestions on how to trouble shoot this please?

Thanks,
Tobin.

keith-packard, 2020-11-14:
(this is a fine place to ask this question, perhaps the 'mu' project would be "more correct', but this is working!)

Perhaps you used the 'master' branch of that repository? That doesn't have the snek patches as it tracks upstream mu's master branch. The branch with the snek changes is the 'snek-merge' branch, or even the snek-debian branch. That's what I used to build the packaged versions I made available.

https://github.com/keith-packard/mu/tree/snek-merge

You should be able to tell that you've got the right version because it will add a 'snek' mode to the Mode menu. If you're running the right version, we'll have to try some more debugging adventures. Maybe it could be a permissions issue?

tcharding, 2020-11-23:
Thanks @keith-packard! Lego robot sessions can be quite spread out at the moment but I'll be in touch with my progress either positive or negative!

Thanks again

keith-packard, 2020-11-23:
Awesome! Have fun with robots. I'm not getting much robot fun this year as our lego classes have been canceled! Hoping to get back to teaching next September.

tcharding, 2021-04-19:
Hi Keith,

Apologies for the outstandingly late response. For future readers, it ended up being caused by incorrect permissions on `/dev/ttyACM0`.

I've had a couple of interesting and fun sessions with my son now, thanks!

closed 2021-04-19 by tcharding