keith-packard/snek #37: while with formatted (string interpolation) print reboots or hangs on Arduino UNO R3

mobluse, 2021-10-17:
I installed Snek 1.7 on two Arduino UNO R3 boards. One original and one clone, but they give the same bug regarding the example with `while` from the tutorial in the documentation at https://sneklang.org/doc/snek.html#_while.

    Welcome to Snek 1.7
    > eeprom.show()
    > eeprom.erase()
    > def f_to_c(F):
    +     return (5/9) * (F - 32)
    + 
    > f_to_c(38)
    3.3333335
    > def f_to_c_table():
    +   F = 0
    +   while F < 100:
    +     C = f_to_c(F)                                                                               
    +     print('%f F = %f C' % (F, C))                                                               
    +     F = F + 10                                                                                  
    +                                                                                                 
    > f_to_c_table()                                                                                  
    0.000000 F = -17.777779 C                                                                         
    10.000000 F = -12.222223 C                                                                        
    20.000000 F = -6.666667 C                                                                         
    30.000000 F = -1.111111 C                                                                         
    40.000000 F = 4.444445 C                                                                          
    50.000000 F = 10.000000 C                                                                         
    60.000000 F = 15.555556 C                                                                         
    70.000000 F = 21.111113 C                                                                         
    Welcome to Snek 1.7                                                                               
    > 

The output stops after 70 F, and the Arduino UNO R3 is reset by itself. Sometimes it hangs and Ctrl+C doesn't work. I believe this is related to the formatted `print` since it works if I replace it with `print(F, C)`. The `for` example works even though that also uses print with string interpolation. 

Flashing went without problems using a Raspberry Pi 4 B 8 GB with Raspberry Pi OS (Raspbian GNU/Linux 10 (buster)):
`/home/pi/.local/share/arduino-1.8.16/hardware/tools/avr/bin/avrdude -C/home/pi/.local/share/arduino-1.8.16/hardware/tools/avr/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:/home/pi/Downloads/snek/snek-uno-1.7.hex:i`

I started with Snek only a few hours ago.

keith-packard, 2021-11-05:
Looks like this example causes a C stack overflow, crashing the interpreter. I've tested a version with a smaller Snek heap, which leaves more space for the stack, and things work fine. See if this works for you!

[snek-uno-1.7.zip](https://github.com/keith-packard/snek/files/7480514/snek-uno-1.7.zip)

closed 2021-11-05 by keith-packard

mobluse, 2021-11-07:
When I connect to the Arduino UNO R3 with `minicom -D /dev/ttyACM0` or press the reset button it shows:
```
<stdin>:51 out of memory
> 
```
And not Welcome to Snek... It does show `Welcome to Snek 1.7` briefly, but then the screen is deleted and the error message is shown instead.

In snekde I get (since it seems to remove Esc):
```
> Welcome to Snek 1.7
[2J[H<stdin>:51 out of memory
> 
```
The `while` example works now, but it's not so nice to be greeted with an error message.

I tried both this version and one that I compiled myself, but get the same result.

Flashing went without problems for both hex-files.

mobluse, 2021-11-07:
`eeprom.erase()` fixed the issue with `out of memory` on reset. There must have been some garbage there.

PR #41 (2021-11-05): Bug fixes
Closes #37 
Closes #40
commit 095da868 (2021-11-05): chips/atmega: Reduce pool on 328p to 900 bytes

This leaves more space for the stack, which is required to run
the f_to_c demo (at least).

Closes #37