keith-packard/snek #42: Snek on Arduino Nano

helmutg, 2021-11-07:
I am wondering whether it would be possible to run snek on a [Arduino Nano](https://www.arduino.cc/en/pmwiki.php?n=Main/ArduinoBoardNano). It is very similar to the Arduino Duemilanove and can be programmed in mostly the same way. As far as I can tell, most Nano board should come with optiboot preinstalled. I attempted flashing `snek-duemilanove-1.7.hex`. To do so, I had to reduce the baud rate passed to `avrdude` to `-b 57600` as that's what the bootloader uses. When booting, it prints `Wel<garbage>` at baud rate 115200. I suppose that not much is missing to make it do something. I had the impression that it does not work well with higher baud rates, so maybe reducing it is all that is necessary.

drawkula, 2021-11-07:
> As far as I can tell, most Nano board should come with optiboot preinstalled.

Please double check that.

[Arduino Products > Arduino Nano # techspecs](https://www.arduino.cc/en/pmwiki.php?n=Main/ArduinoBoardNano#techspecs) mentions:

    | Flash Memory | 32 KB of which 2 KB used by bootloader |

That 'smells' like the Duemilanove boot loader.

helmutg, 2021-11-07:
Wow. So all that needs doing here is updating the documentation to say that Nano works like the Duemilanove. And me updating my bootloader...

drawkula, 2021-11-07:
It is in the docs.

keith-packard, 2021-11-08:
Seems like it would be nice to have instructions for switching to optiboot. I seem to recall an arduino package that you would build and run using the old boot loader that would switch the device to optiboot. Might be nice to at least provide links to that?

drawkula, 2021-11-09:
I only remember [Micronucleus](https://github.com/micronucleus/micronucleus) updates in place via a "normal sketch".

Maybe someone else has the link to in place upgrades to Optiboot?
Is changing the fuses without an ISP possible at all?

commit 1e76bd80 (2021-12-13): doc: Add docs about “big” version for 328 boards

Also mention that the duemilanove version runs on the Arduino Nano.

Closes #42
closed 2021-12-13 by keith-packard