keith-packard/snek #61: build fails with ../../snek-gram.ll:258:5: error: ‘for_depth’ undeclared

fanoush, 2022-09-04:
I am trying to build from source on debian 11 bullseye i386 (old laptop) and aarch64 (raspberry Pi4 with 64bit OS) and on both of them it fails like this
```
make[1]: Entering directory '/home/pi/snek/ports/posix'
lola -DSNEK_SLICE -DSNEK_DICT -o snek-gram.h ../../snek-gram.ll
python3 ../../snek-builtin.py ../../snek-keyword.builtin ../../snek-base.builtin snek-posix.builtin ../../snek-math.builtin ../../snek-input.builtin -o snek-builtin.h
  CC  snek-builtin.o
  CC  snek-code.o
  CC  snek-exec.o
  CC  snek-error.o
  CC  snek-frame.o
  CC  snek-func.o
  CC  snek-lex.o
  CC  snek-list.o
  CC  snek-memory.o
  CC  snek-name.o
  CC  snek-parse.o
In file included from ../../snek-parse.c:219:
../../snek-gram.ll: In function ‘parse’:
../../snek-gram.ll:258:5: error: ‘for_depth’ undeclared (first use in this function)
  258 |     for_depth--;
      |     ^~~~~~~~~
../../snek-gram.ll:258:5: note: each undeclared identifier is reported only once for each function it appears in
In file included from ../../snek-parse.c:219:
../../snek-gram.ll:17:11: error: ‘snek_print_val’ undeclared (first use in this function); did you mean ‘snek_print’?
   17 | command  : @{ snek_print_val = snek_interactive; }@ stat
      |           ^~~~~~~~~~~~~~
      |           snek_print
make[1]: *** [../../snek.defs:163: snek-parse.o] Error 1
make[1]: Leaving directory '/home/pi/snek/ports/posix'
make: *** [Makefile:30: ports/posix/snek] Error 2
```

I can see that mising stuff is declared in C macro PARSE_TOP here https://github.com/keith-packard/snek/blob/main/snek-parse.c#L39 but I don't see the macro being used anywhere so I'm not sure how this is supposed to work. also in generated `ports/posix/snek-gram.h` macro `PARSE_TOP` is not used anywhere.

I installed dependencies including lola as per readme, lola was downloaded as
```
http://deb.debian.org/debian bullseye/main arm64 lola all 1.4-1 [12.0 kB]   
```
I am trying default main branch if that matters.

fanoush, 2022-09-04:
Oh, well, I guess the answer is here https://github.com/keith-packard/lola/commit/0b09e90d17664e5f7391f5b749cfeed1cb488530
Looks like lola version 1.4 in latest debian is too old (?) so the README.md is too optimistic :-(

fanoush, 2022-09-04:
OK `sudo vi $(which lola)` and pasting those three lines there fixed the build.

BTW at first I was guessing gawk and some gnu extension since after second read of readme you mention gawk but it is not in install instructions (and bullseye does not install gawk by default, there is mawk) but that was not it, mawk is probably good enough.

keith-packard, 2022-09-04:
> Oh, well, I guess the answer is here [keith-packard/lola@0b09e90](https://github.com/keith-packard/lola/commit/0b09e90d17664e5f7391f5b749cfeed1cb488530) Looks like lola version 1.4 in latest debian is too old (?) so the README.md is too optimistic :-(

I couldn't find anything in README.md that even mentions the version of lola required; it only talks about installing lola from debian unstable. That's not _wrong_, but it's also not very helpful.

keith-packard, 2022-09-04:
> OK `sudo vi $(which lola)` and pasting those three lines there fixed the build.
>
> BTW at first I was guessing gawk and some gnu extension since after second read of readme you mention gawk but it is [...]

Not having gawk won't stop the programs from being built, but it will prevent the build from completing successfully as the snek build scripts use a GNU extension, `strtonum`, which isn't available in mawk.

PR #62 (2022-09-05): Add some more details about build requirements in README.md
Note that a current version of lola and the GNU awk version (gawk) are
required to build Snek.

Closes #61
closed 2022-09-05 by keith-packard