keith-packard/snek #63: Z80 version of Snek?

mchobby, 2022-09-15:
Hi,
It is just a question about feasibility.

Could the Snek could be compiled for Z80 based system? (having 64K based RAM, 32K ROM at most, 8 bits CPU). 

Thanks,
Dominique

mobluse, 2022-09-15:
I believe the lack of a GCC compatible C compiler for Z80 would be a problem. There exists [Z88DK](https://github.com/z88dk/z88dk) as a C compiler for Z80, but I don't know if that is capable enough. The target system could be a popular CP/M computer or emulator e.g. https://github.com/udo-munk/z80pack.

mchobby, 2022-09-15:
Hi,
Thank for the quick reply.
I'm a big fan of MicroPython and I don't see it running on the Z80... so considering Snek does make sense!
The target would be the RC2014 Pro (from rc2014.co.uk) which is quite popular on Internet.

Presently, I'm just discovering the RC2014 ecosystem, hardware and programming. I want to make some assembler to renew with my young time (when Internet was not existing yet and C64 ressources hard to find).
Considering Python as an alternative to Basic on Z80 systems (RC2014) is really appealing.
RC2014 do support CP/M 2.2 and I though that running Snek on the top of CP/M is the right approach.
C compiler for Z80 are quite old indeed, I will have a look for it but it will be in a serious while (first thing first).

drawkula, 2022-09-16:
> C compiler for Z80 are quite old indeed, I will have a look for it but it will be in a serious while (first thing first).

https://github.com/agn453/HI-TECH-Z80-C still gets updates.

Especially have a look at https://github.com/agn453/HI-TECH-Z80-C/blob/master/overlays/OVREADME.TXT as using overlays may free some more RAM if not all code is needed at every time, e.g. REPL-only stuff versus running a script.

mchobby, 2022-09-16:
> I believe the lack of a GCC compatible C compiler for Z80 would be a problem. There exists [Z88DK](https://github.com/z88dk/z88dk) as a C compiler for Z80, but I don't know if that is capable [...]

From what I know about it, the Z88DK can be used from a Linux machine to compile native Z80 (and probably CP/M based Z80) C and assembler. John's is building a Z80 Retro board from scratch (Hardware, Bios, etc) to boot from SD Card. It also explains how it setup the compiling environment on a Raspberry-Pi. There are also several videos around the CP/M 2.2 implementation.
A must! (See the John's basement channel here  https://www.youtube.com/watch?v=oekucjDcNbA&list=PL3by7evD3F51Cf9QnsAEdgSQ4cz7HQZX5 )

keith-packard, 2022-09-16:
I think any compiler that is c99 compatible will work -- I just did a brief experiment with one of the samd21 targets adding `-std=c99` to the compiler command line. All I had to do was switch the `asm` statements to `__asm__` and it build just fine.

rog77, 2023-05-06:
Would it be possible to run it on this? https://github.com/OLIMEX/AgonLight2

keith-packard, 2023-05-06:
I suspect it could be made to work; finding a C compiler that supports C99 would be the first step.

rog77, 2023-05-06:
I am new to Z80, or at least thinking about doing anything other than playing games on them as a kid, but would any of this help? 

https://sdcc.sourceforge.net/

https://k1.spdns.de/Develop/Projects/zasm/Documentation/z86.htm

https://github.com/z88dk/z88dk/wiki#newlib

Partial C99
https://www.softools.com/scz180.htm

I think these boards are going to be quite popular Pi Hut and Mouser are both selling them.

drawkula, 2023-05-06:
Targetting one brand of board only would be ignoring >99% of the Z80 digitope.

A CP/M version would be nice.
And a Z80 barebone version too.

These 2 variants would cover most of the existing Z80 systems while maybe not using every feature of every board.

rog77, 2023-05-16:
Correct me if I am wrong, but wouldn't there need to be a port for the Z80 in the "chips" part of the code before one could consider targeting a specific board? Is this very difficult to do for someone who knows what they're doing?

Adding in AgonLight afterwards would, as far as I can tell, require inline assembly to interact with the MOS API that handles most I/O, graphics, and storage.

I would have happily had a go at testing compilers myself, but generating a port for the chip is beyond me, getting it to work with the board afterwards seems more achievable for me to try to help with.

Python on a Z80, come on, you know you want to :-))