keith-packard/snek #67: Link to mailinglist returns 403 "Forbidden"

mirko, 2022-12-11:
The link [https://keithp.com/mailman/listinfo/snek](https://keithp.com/mailman/listinfo/snek) on [https://sneklang.org/](https://sneklang.org/) results in a direct HTTP error code 403 ("Forbidden")

keith-packard, 2022-12-11:
Hrm. it's working for me. But, github is probably a friendlier place for bug reports anyways. I need to spend some quality time fixing mailman on that host...

keith-packard, 2022-12-15:
Right, there are spammers out there using mailman subscription scripts to send annoying emails to victims. I had to disable external access to my mailing list web pages as a result. I need to upgrade to mailman3, which (I hope) has this issue solved somehow.