keith-packard/snek #68: Support for small-sized STM32 variants?

mirko, 2022-12-11:
Is there any chance for support of variants of the STM32 family which are too small to run MicroPython, such as the TSSOP20-packaged ones, e.g. STM32F030F4P6*?
The SDK/HAL seems fairly well understood and stable and snek would be a great choice to run these SoCs.
I would have asked on the mailing list, however there seems to be some issue (https://github.com/keith-packard/snek/issues/67)

keith-packard, 2022-12-11:
Yes, this should be possible, even without the SDK (which, iirc, has licensing issues). I've got lots of STMF042 support over in altos (https://git.gag.com/?p=fw/altos;a=summary) which is where the samd21 code for snek started. You'd want parts with at least 32kB of flash and 4kB or more of SRAM. 64kB of flash and 8kB of flash is an awesome size -- you can have all of the snek code and have plenty of space for applications. You might want an off-board SPI flash to store snek source in, but if you've got spare flash in the part you can place it there as well.

mirko, 2022-12-11:
Thanks a lot for the feedback.
Unfortunately the TSSOP20-packaged STM32 variants are rather limited in flash / memory.
At least mouser doesn't list me anything but 16kB flash and 4kB memory.
My use-case would be reading data via ADC, do some math on and write it out via SPI (as SPI slave).
Do you think that's feasible on a STM32F030F4P6TR with above stated limitations?
Also, since there's nothing about stm32 in the docs / git: what would be a good starting point?

mirko, 2022-12-11:
>  At least mouser doesn't list me anything but 16kB flash and 4kB memory.

Ok, at least digikey has TSSOP20-packages with 32kB flash - and even in stock.

keith-packard, 2022-12-11:
I'd encourage you to choose the larger chip, even though it's in a LQFP -- 64kB of flash will be enough to hold most of snek (you might have to skip the math library), and probably leave space in flash to store snek programs themselves, without an external SPI flash part. 32kB is "enough" for the atmega328, but that's with some pretty serious compromises, and on an architecture with a denser ISA. https://www.digikey.com/en/products/detail/stmicroelectronics/STM32G030K8T6/10326689

keith-packard, 2022-12-11:
ooo, they've got this one with 64kB in a tssop-20 package: https://www.digikey.com/en/products/detail/stmicroelectronics/STM32G031F8P6/10326693