keith-packard/snek #78: RISC-V support like CH32V003 MCU

dfsxn, 2023-02-12:
Hi,

Last year it was launched to the market the MCU CH32V003, which has the following features:

[image]

CH32V003 series is based on QingKe RISC-V2A core design of industrial-grade general-purpose microcontroller, support 48MHz system main frequency, with wide voltage, 1-wire serial debug interface, low-power consumption, ultra-small package, etc. CH32V003 series built-in a group of DMA controller, a group of 10-bit ADC, a group of op-amp comparators, multiple timers and standard communication interfaces USART, I2C, SPI, etc.

http://www.wch-ic.com/products/CH32V003.html

Resources are available on GitHub and GitLab:

* https://github.com/openwch/ch32v003
* https://gitlab.com/eta-for-ci/ch32v307-gnumake
* https://www.youtube.com/watch?v=L9Wrv7nW-S8

Even there's also a WCH Kicad Library on the way:

* https://github.com/Taoyukai/wch_kicad_library
* https://github.com/sad-electronics/wch-kicad-lbr

Board available on the market:

1. https://www.aliexpress.com/item/1005004895791296.html

It would be beautiful have Snek Python support for CH32V003 MCU.

Thank you.

Kind regards,

keith-packard, 2023-02-12:
That looks a bit too small for snek -- snek needs at least 32kB of flash for even the smallest configuration.

keith-packard, 2025-08-20:
cool. I ordered a board to see what it's like.