keith-packard/snek #84: compilation fails Picolibc semihosting

sanyaade-teachings, 2023-08-23:
Hi All,

Good day to everyone and many thanks for snek and all your amazing efforts
I am having small issue getting snek to build fully on Ubuntu 20.04 on WSL. The build fails on semihosting as shown below:

[image]

I followed all the install steps as detailed on Snek GitHub pages.
 
Please help for tips and directions!
Hear from you soon!

God blesses!!!

Regards

keith-packard, 2023-08-23:
You'll need a newer version of picolibc for this to work -- version 1.7.5 or newer, which means upgrading to ubuntu 22.10 or later. Good luck!

sanyaade-teachings, 2023-08-24:
> You'll need a newer version of picolibc for this to work -- version 1.7.5 or newer, which means upgrading to ubuntu 22.10 or later. Good luck!

Many thanks!
Following your direction, 
I upgrade the Ubuntu distro to version 23.10
build snek again. It all went well.
[image]

Many thanks!
Next is to learn the basic steps requiring hardware porting with snek.
Many thanks!

God blesses!!!

Regards

closed 2023-08-24 by sanyaade-teachings