keith-packard/snek #85: Size Deails please

PythonLinks, 2024-05-26:
This looks really interesting. 
I am considering mentioning it in a talk I am giving tomorrow. 
Slide Page 11. 
https://pythonlinks.info/presentations/forth/YUG27May2024.pdf

The interesting metric is how much RAM and how much FLASH is required.   When I do an 

ls -lh  *.hex

On your distribution file, I get 90+ K hex characters, which tells me that you require 45KBytes of Flash.  Is that correct?  How much RAM do you take to just load the system, but not execute any user code?  

Forgive me, I am always reluctant to fire up downloaded code on my linux box.  In any case, it would help make your web site a bit better.  You may also want to quote the numbers for MicroPython, given on the page that that page 11 links to. 

If I make any changes to that PDF, the page number may change by 1 or 2.

keith-packard, 2024-05-26:
The size depends upon the target and the build configuration. Snek has a bunch of functionality which can be enabled or disabled, including numerous builtin functions and even core language functionality like dictionaries. The existing targets range from just under 32kB through about 64kB of ROM. Aside from the heap, snek uses about 1kB of RAM for various internal housekeeping tasks. That leaves a 1kB heap on the ATMega 328p.

The biggest reason snek uses so much flash is that it includes a full compiler, a garbage collector and runtime to interpret the bytecodes generated by the compiler. I've thought about how a pure interpreter (ala lisp or basic) would work, but haven't ever managed to come up with a credible design.

PythonLinks, 2024-05-26:
Thank you for the fast reply. 
I updated that page. 
[image]