keith-packard/snek #88: All modules should automatically be from module import *

mobluse, 2024-06-16:
It would be easier if `from math import *` was used because it would save people from having to type `math.` before every math function. All modules are auto imported, and I suggest all modules would also be auto `from`-imported. That means you could use all functions without a module name prefix. You could still use the fully qualified name if you want or need to. If one name shadows another you have to use the fully qualified name except for the last `from`-imported module. 

The modules could be auto `from`-imported in an order so that the most used modules are imported last.

`from ANYTHING import *` should be ignored like `import ANYTHING` is ignored now.

I think this would be helpful for beginners, since you don't have to type many module prefixes. This means Snek would be more competitive as a first computer language. This would also make Snek programs shorter in flash memory.

E.g. `stdscr.refresh()` would still need `stdscr.` since `stdscr` is an object and not a module.

keith-packard, 2024-06-17:
snek doesn't actually support python modules, the whole hack with the math functions is that they really are named `math.sin` et al. `.` is part of the name, it's not separate at all. Maybe those functions should also be available without the `math.` prefix as well, so that you could pretend that `from math import *` did what you expected?

mobluse, 2024-06-22:
I understand that the modules are faked, but I think it should work as if this had been in a Python3 file before the rest of the program:

```
import eeprom
import tone
import curses
import time
import random
import math
from eeprom import *
from tone import *
from curses import *
from time import *
from random import *
from math import *
```

I.e. the most common modules should be from-imported last in order to override earlier names.

stdscr is an auto created object in the Snek case.

GPIO functions, e.g. talktio() and on(), are moduleless and temperature() also.
Is it reset() or eeprom.reset()? See https://sneklang.org/doc/snek.html#_reset .

merwok, 2024-06-22:
62 names (from the math module) seem a bit much, IMHO it’s cleaner and closer to Python to have only `math.x` names

mobluse, 2024-06-26:
It's standard on graphing calculators with Python to from-import `math` and other modules. It makes programming easier and programs shorter.

mobluse, 2024-10-24:
An alternative could be to allow shorter module names. It would be as if the following had been done:
```
import eeprom as e
import tone as t
import curses as c
import time as t
import random as r
import math as m
```
Then one wouldn't need to type a lot and one could fit larger programs in flash.

```
$ python
Python 3.11.2 (main, Aug 26 2024, 07:20:54) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import math
>>> m=math
>>> m.atan(1)*4
3.141592653589793
>>> 
$ snek
Welcome to Snek version 1.9
> import math
> m=math
<stdin>:2 undefined: math
> m.atan(1)*4
<stdin>:3 undefined: m.atan
> atan=math.atan
> atan(1)*4
3.14159274
>
```

keith-packard, 2024-10-24:
I'd probably remove `time.` and `random.` as well. Unsure about the others.

mobluse, 2024-10-25:
I believe this would be an improvement since math, random, and time mostly uses `from` module `import` functions. It would also be easier to type and store in eeprom. This change would break existing Snek programs, but one could write a sed script to convert them.

keith-packard, 2024-10-28:
Check out PR #99 and see what you think. I've left curses, stdscr and eeprom out of this patch because they're much less common. The eeprom functions can't be changed without also fixing the mu editor upstream.

mobluse, 2024-10-29:
It works as expected from my small tests e.g. using this program:
https://github.com/mobluse/python82-scripts/blob/main/flappbat_s.py
Those ending in `_s.py` works in the coming Snek.
I compiled using (since I only compile for Linux and Arduino UNO):
```
sudo apt install lola gcc-avr avr-libc python3-serial gawk libreadline-dev asciidoctor ruby-asciidoctor-pdf
cd
git clone https://github.com/keith-packard/snek/
cd snek/ports/posix/
make clean
git fetch origin pull/99/head:BRANCH_NAME
git switch BRANCH_NAME
make
sudo make install
```
I have not yet tested on Arduino UNO.

keith-packard, 2024-10-29:
Thanks for testing. I think I'll go see if I can add an option to leave the old names in place while also providing the new names on targets large enough for that. Tiny targets will want to save the space, I suspect.